ADVRHumanoids/CartesianInterface #2: Feature: Velocity limit when publishing references

erolley-parnell, 2018-05-09:
Is there any plan in the pipeline to make a velocity limiting option? For example, where the reference is published and the joint will move towards the PoseStamped location at a specified velocity until another message is received?

alaurenzi, 2018-05-09:
Hi! For the near future, no. But it is a good suggestion

alaurenzi, 2018-07-23:
Sort-of-implemented in devel branch. Needs testing though. Basically you can set velocity/acceleration limits by changing the following parameters 

```
/xbotcore/cartesian/task_name/max_acceleration_angular
/xbotcore/cartesian/task_name/max_acceleration_linear
/xbotcore/cartesian/task_name/max_velocity_angular
/xbotcore/cartesian/task_name/max_velocity_linear
```
and **then call**  the service `/xbotcore/cartesian/update_velocity_limits`