ADVRHumanoids/CartesianInterface #4: Can't find XBotInterface/ConfigOptions.h

yangweiyou, 2018-06-28:
I want to try CartesianInterface, but it showed when making:

cartesian_interface/utils/LoadConfig.h:4:41: fatal error: XBotInterface/ConfigOptions.h: No such file or directory

I update the XBotInterface module, still no this file. I think it may exist in an old commit.

EnricoMingo, 2018-06-28:
Hi Yangwei,
You should put: 
- Cartesian Interface to config_refactor branch
-XBotInterface to devel
- Model InterfaceRBDL to devel
- opensot to devel
-RobotInterfaceROS to devel 

It should compile everything in this way.
Inviato da iPhone

alaurenzi, 2018-06-28:
Yes Yangwei sorry, the master branch now supports our latest devel branches from XBotCore (to be merged soon to master)

yangweiyou, 2018-06-29:
Thank you, guys! I will try it. In the beginning, I tried to find the missing head file in the devel branch of XBotInterface but didn't succeed. Maybe I missed somewhere.

closed 2018-07-03 by yangweiyou