ADVRHumanoids/CartesianInterface #97: Porting bindings to ROS2

torydebra, 2025-05-08:
Hi, 

I need some help for porting the bindings in ROS2. At the moment I am stuck at the [RosClient.h](https://github.com/ADVRHumanoids/CartesianInterface/blob/3.0-devel/include/cartesian_interface/ros/RosClient.h), and the relative [RosImpl.cpp](https://github.com/ADVRHumanoids/CartesianInterface/blob/3.0-devel/src/ros/RosImpl.cpp) which uses this header.

For example, there is this Initializer which utilizes a ros::callbackqueue, and I am not sure the logic to "convert" it to ROS2, since there is no such object in ros2 as far as I understood

https://github.com/ADVRHumanoids/CartesianInterface/blob/64933f220ecc792f799c3c2668c05cf93f2f4371/include/cartesian_interface/ros/RosClient.h#L15C1-L27C1

@alaurenzi @liesrock @EdoardoRomiti

liesrock, 2025-05-08:
check this!

https://github.com/ADVRHumanoids/xbot2/blob/ros2/ros2/src/ros2_support.cpp

torydebra, 2025-05-08:
Ok, I will check it, devs are on the `ros2-magician-binding-wip` branch

torydebra, 2025-05-26:
done I would say

closed 2025-05-26 by torydebra